lategoodbye/rpi-zero #21: Upstreaming of raspberrypi patches

lategoodbye, 2017-11-04:
Currently the foundation repository contains a lot of patches against a stock kernel. This ticket should document the status of the current upstream work:

bcm2835-aux: Add aux interrupt controller
https://github.com/raspberrypi/linux/pull/2236/commits/401b4baa052b64f79a82472cf81aef5666a8304e
Status: Needs fixing of the last comments

bcm2835-gpio-exp: Driver for GPIO expander via mailbox service
https://github.com/raspberrypi/linux/pull/2236/commits/32b813c05b3747a538360979f190e62bb85b071b
Status: Handled by Dave Stevenson

BCM2835_DT: Fix I2S register map
https://github.com/raspberrypi/linux/pull/2236/commits/304e6c3e84723df32bc73a63902bcc94ba348358
Status: Needs fixing

bcm2835-rng: Avoid initialising if already enabled
https://github.com/raspberrypi/linux/pull/2236/commits/11de561503e681ee9207ff460f4516957d699b3a
Status: Delayed because of https://marc.info/?l=linux-arm-kernel&m=150958473206921&w=2

irq-bcm2836: Avoid "Invalid trigger warning"
https://github.com/raspberrypi/linux/pull/2236/commits/060f2862ae64c115b5f5a792581b72201d5d5840

amba-pl011: Report AUTOCTS capability to framework 
https://github.com/raspberrypi/linux/commit/0ec723f8c6990f8ab740c54796d85f0474e77b88
Status: Replaced by https://marc.info/?l=linux-serial&m=150891880104614&w=4

amba_pl011: Insert mb() for correct FIFO handling
https://github.com/raspberrypi/linux/commit/b84ce141b9306937f96aaec682e93b0d2058dce9
Status: unknown

nullr0ute, 2018-08-28:
Does upstreaming also cover patches for official Foundation HATS like the sense HAT too?

lategoodbye, 2018-08-29:
This issue is more intended for minor general platform patches which hasn't been upstreamed yet. Overlays and especially the sense hat with its mfd driver is a different thing. But everyone is invited to send patches.