lategoodbye/rpi-zero #23: VCHIQ: data corruption during functional vchiq_test

lategoodbye, 2017-11-22:
Runnung vchiq_test -f 10 on RPI 3 (arm64, defconfig, linux-next-20171122) shows the following output:
```
Functional test - iters:10
======== iteration 1 ========
0041e060,10: 10 <-> 00
0041e061,11: 11 <-> 00
0041e062,12: 12 <-> 00
0041e063,13: 13 <-> 00
0041e064,14: 14 <-> 00
0041e065,15: 15 <-> 00
0041e066,16: 16 <-> 00
0041e067,17: 17 <-> 00
0041e068,18: 18 <-> 00
0041e069,19: 19 <-> 00
0041e06a,1a: 1a <-> 00
0041e06b,1b: 1b <-> 00
0041e06c,1c: 1c <-> 00
0041e06d,1d: 1d <-> 00
0041e06e,1e: 1e <-> 00
0041e06f,1f: 1f <-> 00
0041e070,20: 20 <-> f0
0041e071,21: 21 <-> f1
0041e072,22: 22 <-> f2
0041e073,23: 23 <-> f3
0041e074,24: 24 <-> f4
0041e075,25: 25 <-> f5
0041e076,26: 26 <-> f6
0041e077,27: 27 <-> f7
0041e078,28: 28 <-> f8
0041e079,29: 29 <-> f9
0041e07a,2a: 2a <-> fa
0041e07b,2b: 2b <-> fb
0041e07c,2c: 2c <-> fc
0041e07d,2d: 2d <-> fd
0041e07e,2e: 2e <-> fe
0041e07f,2f: 2f <-> ff
mem_check failed - buffer 41e050, size 1024
0041e840,3f0: f0 <-> 00
0041e841,3f1: f1 <-> 00
0041e842,3f2: f2 <-> 00
0041e843,3f3: f3 <-> 00
0041e844,3f4: f4 <-> 00
0041e845,3f5: f5 <-> 00
0041e846,3f6: f6 <-> 00
0041e847,3f7: f7 <-> 00
0041e848,3f8: f8 <-> 00
0041e849,3f9: f9 <-> 00
0041e84a,3fa: fa <-> 00
0041e84b,3fb: fb <-> 00
0041e84c,3fc: fc <-> 00
0041e84d,3fd: fd <-> 00
0041e84e,3fe: fe <-> 00
0041e84f,3ff: ff <-> 00
mem_check failed - buffer 41e450, size 1024
Testing bulk transfer for alignment.
vchiq_test: 72: Data corrupted at 1-1 (datalen 3, align 0, server_align 0) -> 00
vchiq_test: 891: func_data_test(service, size, align & 31, srvr_align & 31) != VCHIQ_SUCCESS
```

lategoodbye, 2017-12-13:
I tested today 4.15-rc2 and it's also affected by this issue.

sergey-suloev, 2018-01-17:
@lategoodbye any news on that ? got the same issue with latest 4.15-rc8

lategoodbye, 2018-01-19:
Unfortunately not, i'm busy with other BCM2835 parts. I suggest to bisecting this issue. bcm2835_defconfig should be a good starting point.

closed 2018-11-18 by lategoodbye