lategoodbye/rpi-zero #32: cpufreq: Implement cpufreq support for Raspberry Pi

lategoodbye, 2018-05-23:
In a recent [discuss](http://lists.infradead.org/pipermail/linux-rpi-kernel/2018-April/007515.html) someone requested for cpufreq support.

There are at least two options:
- implement a bogus clock driver and use cpufreq-dt
- implement a separate cpufreq-bcm2835

lategoodbye, 2018-05-23:
Unfortunately me too, even after implementing recalc_rate with RPI_FIRMWARE_GET_CLOCK_MEASURED.

nullr0ute, 2018-06-04:
By just catting /sys/devices/system/cpu/cpufreq/policy0/scaling_cur_freq in brief testing I've seen it touch on 900 on it's way between 600/1200.

The patch looks to be just for the original RPi3, is there just more DT bits required for the 2/3+ (and presumably the various 0/1 devices)?

lategoodbye, 2018-06-04:
This isn't a topic which isn't fully matured yet. So i'm not interested in pushing a quick shot for 4.19.

Currently i'm not sure which approach is better. The DT has the disadvantage that we need to model the OPPs on board level instead of SoC. That's why i started with RPI 3.

nullr0ute, 2018-06-04:
I realise, I'm just playing to see if it's something that would be usable in the interim across the devices I need to support.

[0001-add-1.4-ghz-OPP-for-the-3B.patch.txt](https://github.com/lategoodbye/rpi-zero/files/2067395/0001-add-1.4-ghz-OPP-for-the-3B.patch.txt)

So this appears to work on the 3B+
# cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available
600000 900000 1200000 1400000 

No idea if the 3B+ supports 1.2 as well as 1.4 though. I suppose similar would work for the other devices

lategoodbye, 2018-06-04:
AFAIK firmware supports arbitrary values. So this should be okay.

But this was only the easy part, because all driver connected to the VPU clock must handle clock changes now.

nullr0ute, 2018-06-04:
I have no doubts it was the easy part, is there details of which drivers are affected?

The best I've managed to find on the freqs supported seems to be this section in the wikipedia docs:
https://en.wikipedia.org/wiki/Raspberry_Pi#Overclocking

sergey-suloev, 2018-06-04:
If every driver needed to take CPU clock change into account that would be a hell. I am not sure how it is implemented in Rpi but normally it should be handled in only one driver, like ccu. Correct me if I am wrong.

sergey-suloev, 2018-06-04:
@lategoodbye And what about voltages ? Usually OPP table contains freq + voltage, does Rpi need voltage change ?

lategoodbye, 2018-06-04:
Please look at all driver which uses BCM2835_CLOCK_VPU for the notification changes:
https://elixir.bootlin.com/linux/latest/source/arch/arm/boot/dts/bcm283x.dtsi

The VC4 firmware takes care of the voltages.

sergey-suloev, 2018-06-04:
@lategoodbye  Is there a way to change(or create) only one , let's cal it "hub driver", and what would it cost ?

lategoodbye, 2018-06-04:
Sorry, i don't understand question / intension?

sergey-suloev, 2018-06-04:
@lategoodbye  my point is that changing every driver in order to integrate OPP table doesn't make too much sense.

lategoodbye, 2018-06-04:
I think there is a misunderstanding. You don't need to implement OPP table handling in every driver. But a lot of driver uses the VPU as clock input with its clock divider. So in case the VPU clock changes the clock divider should be adjusted, too. Otherwise the resulting frequency is wrong. Usual the divider settings are claimed by the relevant driver, so in case of a VPU clock change is makes sense to handle it by the relevant driver.

nullr0ute, 2018-06-04:
@lategoodbye so that was my understanding around the drivers understanding that a clock would possibly change under them as opposed to being static. My query was is can we grep the DT or a string in drivers for which upstream drivers might be affected to work out what might be affected?

lategoodbye, 2018-06-04:
As i wrote above grep for BCM2835_CLOCK_VPU within the DT.

sergey-suloev, 2018-06-04:
@lategoodbye That's what I am talking about too. Can we have a special driver that will provide stable clocks for those drivers no matter if CPU clock changes or not?

nullr0ute, 2018-08-31:
So having been running this for a little bit in Fedora one thing to note here when running something that I've noticed when running something that taxes the gpu too such as gnome on wayland we get shutdowns/lockups/poweroffs. I need to investigate more and get more debug.

The end driver might want to tie into the thermal framework and possibly the hwmon for undervoltage and use those for scaling decisions too.

lategoodbye, 2018-08-31:
@nullr0ute Do you mean Sergey's patch?
Which governor did you use?

sergey-suloev, 2018-09-02:
@nullr0ute  can it be due to your power supply limits ?

nullr0ute, 2018-09-02:
@sergey-suloev I don't believe so, it's a good quality Anker PSU, I will be investigating more this week. I think it's more thermal than power TBH but I wasn't remotely connected in to see any logs at the time. The RPi3 doesn't have a heatsink on it.

nullr0ute, 2018-09-04:
For reference with more testing, it's thermal, I get the following:
kernel: thermal thermal_zone0: critical temperature reached (80 C), shutting down

lategoodbye, 2018-09-19:
I want to point out to this commit 0fe4d2181cc4cb3eba303c0e03f878d2558d0f3a

In the lack of a cooling device this was left empty. But with a cpufreq driver we have such a cooling device.

sergey-suloev, 2018-09-19:
@lategoodbye could you provide the code that should be added, please ?

lategoodbye, 2018-09-19:
It's not code, it's device tree settings. Yes, later

lategoodbye, 2018-09-26:
Based on Peter's cpufreq patch i made a [patch](https://gist.github.com/lategoodbye/0f304a5f57ccc2b31bffb2466e2e1818) to use CPU0 of RPI 3 as cooling device.

Disclaimer: Only compile tested

nullr0ute, 2018-09-26:
Initial feedback, the cooling device shows up tmon and P trip point is there. Will test the actual trips shortly

nullr0ute, 2018-09-26:
BIG :+1: It seems to work pretty well. I've not managed to have my original RPi3 shut down due to thermal. Remotely running tmon and following the current freq in /sys/devices/system/cpu/cpufreq/policy0/ I've not managed to trip thermal in basic testing. It has got above the 70 P but managed to keep it well below Critical 80 and seems to work pretty well :+1:

lategoodbye, 2018-09-27:
Thanks for testing. Maybe we should add a hysteresis of 2 degree.

sergey-suloev, 2018-09-27:
@lategoodbye Stefan, is it possible to make Rpi3 work on 900MHz ?

nullr0ute, 2018-09-27:
@sergey-suloev in my testing I see it running at 900mhz on occasion, it moves between 600/900/1200

sergey-suloev, 2018-09-27:
@nullr0ute  I can't make it work on 900 when I set manually with userspace gov. The real freq is always 1200.

lategoodbye, 2018-09-27:
I think the 900 MHz which Peter is seeing is only internal in the cpufreq driver.

nullr0ute, 2018-09-27:
I'm looking in /sys/devices/system/cpu/cpufreq/policy0/ is there other places that are more accurate?

sergey-suloev, 2018-09-27:
@nullr0ute  can you make it 900 manually ?

lategoodbye, 2018-09-27:
@nullr0ute 
AFAIK there is no chance to fetch actual core frequency via mailbox.

The more accurate way should be:
`vcgencmd measure_clock arm`

sergey-suloev, 2018-09-27:
no matter how you fetch it, on my Rpi3 it is always 1200  after I set it manually to 900

nullr0ute, 2018-09-27:
@lategoodbye unfortunately (last I looked at least) that tool isn't open source and doesn't work on aarch64 either

lategoodbye, 2018-10-13:
@nullr0ute Recently i tested Fedora Workstation 29 Beta 1.5 on my RPi 3 B+. The whole system periodically freeze completely (no USB power on keyboard, no reaction on screen). After blacklisting cpufreq_dt the system becomes usable (including debug UART). Just a warning cherry-picking just this cpufreq patch is very dangerous and can leads to data loss e.g. on sdhost interface.

ggardet, 2019-03-14:
What is the current status? Is there any upstream effort on going?

lategoodbye, 2019-03-14:
Currently not. The first step would be to adapt the relevant driver (sdhost, i2c, aux, spi, dpi) for dynamic clock changes.

lategoodbye, 2019-03-23:
Okay, here is a untested patch to dynamically adjust AUX UART divisor:
https://gist.github.com/lategoodbye/5f4eea59e82c310519adf76dc095fe2e

Be aware this currently wont work with the implementation of clk-bcm2835, because it doesn't take VPU clock changes / Turbo mode into account.

nullr0ute, 2019-03-25:
I wonder if the OPP framework is one way to achieve this, seems this series for qcom is similar to what needs to be achieved.

https://lists.freedesktop.org/archives/dri-devel/2019-March/211321.html

lategoodbye, 2019-04-03:
First RFC series by Nicolas Saenz Julienne:
http://lists.infradead.org/pipermail/linux-rpi-kernel/2019-April/008634.html

lategoodbye, 2019-06-04:
Second RFC series by Nicolas Saenz Julienne:
https://patchwork.kernel.org/cover/10950865/

danieleftodi, 2019-06-04:
> Second RFC series by Nicolas Saenz Julienne:
> https://patchwork.kernel.org/cover/10950865/

So, where might one get a hold of this patch to test? I have a RPi 3B+ running 5.1.6-1 on Arch Linux Aarch64 :-) 

Willing to try it out, and see how it'll go.

lategoodbye, 2019-06-04:
Click "series" on the right and you should get the whole patch series. Be aware this isn't the final version yet.

vianpl, 2019-06-04:
@danieleftodi If you're willing to be patient, I'm planning on posting a new version of the patch series tomorrow. I'll appreciate the testing :wink: .

danieleftodi, 2019-06-04:
> @danieleftodi If you're willing to be patient, I'm planning on posting a new version of the patch series tomorrow. I'll appreciate the testing 😉 .

Sure! ... np. just post a link when you're done, and I'll give it a try.

lategoodbye, 2019-06-04:
@danieleftodi I think it's okay if i post the link to the new patch series:
https://patchwork.kernel.org/patch/10975841/

After applying the patch you still need to enable the CONFIG_ARM_RASPBERRYPI_CPUFREQ in the kernel config.

danieleftodi, 2019-06-05:
@lategoodbye pressing "series" is not recommended :-) it downloads all the e-mail text and patches in one "text dump" :-) 

One has to go to each series and download each one by pressing "diff". 

No biggie, as one can cat $i > one-patch.diff later :-) 

[image]

lategoodbye, 2019-06-05:
Sure you can make everything complicated or just 
`git am series.patch` in the linux repo.

danieleftodi, 2019-06-05:
@lategoodbye  you learn something new every day 🤓

yeah, that makes it much easier.

lategoodbye, 2019-06-08:
The latest and bugfixed version V2 can be found here:
https://patchwork.kernel.org/patch/10979695/

lategoodbye, 2019-06-18:
This is hopefully the last version V4, which has been acked by myself:
https://patchwork.kernel.org/patch/10990631/

nullr0ute, 2019-06-18:
I've been testing v4 of this series on the rpi2/rpi3b/3b+/3a+ (both aarch64 and armv7 mode on the 3 series) and it seems to be working well.

sergey-suloev, 2019-06-18:
@vianpl  thank you very much for your work

vianpl, 2019-06-18:
@nullr0ute Thanks for testing!
@sergey-suloev My pleasure :smile:

closed 2019-10-12 by lategoodbye