lategoodbye/rpi-zero #48: Upstream CM4 and CMIO4

lategoodbye, 2020-10-21:
The new CM4 is very similiar to the Raspberry Pi 4 B

Differences:
- 2 full size HDMI ports
- No VIA Labs VL805 USB 3.0 Host controller (XHCI)
- eMMC is connected via 8 data lines instead of 4
- Dual camera interfaces
- external wifi antenna
- builtin RTC

TODO:
- check Wifi chip

lategoodbye, 2021-07-19:
Here is a first draft:
https://github.com/lategoodbye/rpi-zero/tree/v5.14-cm4-v1

CyrilBrulebois, 2021-07-20:
Oh wow, very nice!

I've tried to cross-build a `linux-image` package from that tree, using the `.config` from the latest Debian kernel (based on 5.10 for now, being stabilized for the upcoming Debian 11 release), using `oldconfig` and picking all default choices for new options.

I've deployed the resulting package in a Debian environment running on a regular Pi and the machine didn't manage to reboot into the new kernel. I suppose there are a lot of changes to catch up with between 5.10 and 5.14, so I've decided to test things a little more manually: I copied the generated `bcm2711-rpi-cm4-io.dtb` file manually in the first partition of a regular Debian Pi image (as built thanks to https://salsa.debian.org/raspi-team/image-specs), both under that name and under the name that can be found in the official Raspberry OS images: `bcm2711-rpi-cm4.dtb`; the said image uses a 5.10.46 kernel, and the `arm64` architecture.

Two hardware environments, and rather good results overall:

- one regular CM4: deployed using USB slave (a jumper is required to enable it, as opposed to the CM3 which advertised a jumper but didn't actually require using it): it boots fully, and enables Ethernet and USB at least;
- one CM4 Lite (no eMMC → using an SD card): it doesn't boot fully; I haven't found anything looking like storage under `/dev` (and `lsblk` doesn't find anything either), which explains why I'm dropped into a shell from the initramfs. I suppose this shouldn't be due to missing modules since the same image is able to boot a Pi 4, from the same SD card.

I'm assuming you're testing with a 32-bit environment, and I'm therefore suggesting the following patch to have the DTB built for arm64 as well. :) → [for-stefan-0001-arm64-dts-broadcom-Add-reference-to-Compute-Module-4.patch.txt](https://github.com/lategoodbye/rpi-zero/files/6844695/for-stefan-0001-arm64-dts-broadcom-Add-reference-to-Compute-Module-4.patch.txt)

Thanks, that looks very promising!

lategoodbye, 2021-07-20:
@CyrilBrulebois Thanks again for the hardware.

Yes, as written above this is just a draft. It has only been tested with a CM4 (not a CM4L) and only 32 bit Raspbian. The DTB has been optimized for eMMC, which should provide HS200 instead of DDR52. So be careful about your data it's possible that this speed is to for the hardware. It took me 2 days to figure out that the firmware was patching the MDIO address of the ethernet PHY. So i wanted to provide the first version which probes ethernet (also untested).

Until i found a smart solution, i could provide a DTS for the CM4 Lite which should boot properly.

lategoodbye, 2021-07-26:
Here is a the second version:
https://github.com/lategoodbye/rpi-zero/tree/v5.14-cm4-v2

nullr0ute, 2021-07-26:
> Here is a the second version:
> https://github.com/lategoodbye/rpi-zero/tree/v5.14-cm4-v2

Has that been posted to mailing lists?

lategoodbye, 2021-07-28:
Not yet. As usual i will post a link in case i send it to the mailing list.

lategoodbye, 2021-08-01:
Okay, here is the third version:
https://github.com/lategoodbye/rpi-zero/tree/v5.14-cm4-v3

which has been sent to the mailing list:
https://marc.info/?l=linux-arm-kernel&m=162782110325813&w=2

lategoodbye, 2021-08-07:
Next round:
https://github.com/lategoodbye/rpi-zero/tree/v5.14-cm4-v4

which has also been sent to the mailing list:
https://marc.info/?l=linux-arm-kernel&m=162833462918135&w=2

lategoodbye, 2021-09-10:
I will close this, since it has been applied to linux-next.

closed 2021-09-10 by lategoodbye