lategoodbye/rpi-zero #51: Broadcom genet fails to attach ethernet PHY on Raspberry Pi 4B sometimes

starnight, 2021-06-21:
I notice the Broadcom genet fails to attach ethernet PHY on Raspberry Pi 4B sometimes (more than 50%) when I test kernel 5.13-rc6, also tested 5.11.

It shows errors in the kernel message:
```
[   10.615778] bcmgenet fd580000.ethernet: GENET 5.0 EPHY: 0x0000
...
[   11.585741] could not attach to PHY
[   11.585790] bcmgenet fd580000.ethernet eth0: failed to connect to PHY
[   11.615180] uart-pl011 fe201000.serial: no DMA platform data
[   11.623193] libphy: bcmgenet MII bus: probed
...
[   11.814783] unimac-mdio unimac-mdio.-19: Broadcom UniMAC MDIO bus
```
The [full dmesg](https://bugzilla.kernel.org/attachment.cgi?id=297443).

I also filed this bug to kernel's bugzilla as [bug 213485](https://bugzilla.kernel.org/show_bug.cgi?id=213485).

According to the diagnostics in [bug 213485](https://bugzilla.kernel.org/show_bug.cgi?id=213485), I found the **unimac-mdio** comes too late for **genet**, or system is too late to load **mdio-bcm-unimac** module.

lategoodbye, 2021-06-21:
@starnight Thanks for reporting this issue. I'm having trouble to reproduce this issue, so i need more information about your setup:

What RPi 4 bootloader version are you using (vcgencmd bootloader_version)?
What distribution do you use?
What filesystem type is your rootfs?
What is connected to your Raspberry Pi 4 (HDMI, USB)?

starnight, 2021-06-21:
> @starnight Thanks for reporting this issue. I'm having trouble to reproduce this issue, so i need more information about your setup:
> 
> What RPi 4 bootloader version are you using (vcgencmd bootloader_version)?

The firmwares are from RPi OS repository at [tag 1.20210527](https://github.com/raspberrypi/firmware/tree/1.20210527/boot).  But, I make it boot U-Boot at [tag v2021.04](https://source.denx.de/u-boot/u-boot/-/tree/v2021.04), then boot mainline kernel (aarch64).

> What distribution do you use?

[Endless OS](https://endlessos.com/download/)

> What filesystem type is your rootfs?

EXT4

> What is connected to your Raspberry Pi 4 (HDMI, USB)?

Both HDMI and USB

lategoodbye, 2021-06-21:
@starnight Could you please apply this [patch](https://gist.github.com/lategoodbye/b962e636543803fedd9cc59028fe6b44), reproduce the issue and provide the dmesg?

lategoodbye, 2021-06-21:
@starnight Please don't provide such ugly workaround patches to the netdev mailing list without helpful context. The results of my former request are more helpful.

lategoodbye, 2021-06-22:
I think, i can close this because the discussion continues here:
https://www.spinics.net/lists/netdev/msg749822.html

closed 2021-06-22 by lategoodbye